microsoft/pxt-chibitronics #190: Request: timer function

nataliefreed, 2018-02-15:
It would be super helpful to have a time elapsed block, similar to Arduino's millis(), or Scratch's 'timer' and 'reset timer' blocks in the sensing category. Ideally it would go down at least to milliseconds - either in seconds as a floating point number, or in milliseconds.

Apologies if I've missed something that already exists. Thank you!

abchatra, 2018-02-15:
@nataliefreed In Circuit Playground Express editor we have  "Get the number of milliseconds of time passed since the board was turned on."  block. See https://makecode.adafruit.com/reference/control/millis

Is that what you need?

nataliefreed, 2018-02-15:
Yes! That's it.
I do like that Scratch also has a 'reset timer' block so you can make it start again at 0 (math with millis is hard for newbies to understand). But millis alone would already be hugely helpful.