microsoft/pxt-chibitronics #191: analog read mapping is not 0-100?

nataliefreed, 2018-02-20:
Analog readings on voltages from 0 to 3V appear to be mapped to 0-400(ish) rather than 0-100. For example, if I run the code below to print analog read values to the Scope, a 3 volt reading (+3V pin connected to pin 0) shows up as 401, rather than 100 as I would have expected. I've tried this with both release and beta versions. Am I missing something or could this be a bug? Thanks!

[image]

[image]

[image]

pelikhan, 2018-02-20:
@bunnie  When is the range of data returned by ``::analogRead``?

xobs, 2018-02-21:
We follow the Arduino convention and default to 10 bits (0..1024), but support 12 bit resolution (0..4096).  A function exists ::analogReadResolution(int bits) to change the resolution, which could be called at the start of the program, or you could divide everything by four in ts.

pelikhan, 2018-02-21:
Same convention here so we’re missing / 4.

samelhusseini, 2018-03-01:
@xobs I've followed this through and it looks like our assumption that the OS ::analogRead function is returning a value between 0 and 255. 
I want to double check on a couple of assumptions to make sure we're consistent. 
it sounds like from your previous comment that the value returned by the OS analogRead is between 0 and 1024. That's okay I can update that assumption from our end, but what about analogWrite. Does it expect a value between 0 and 1024 or does is it inconsistent and expects a value between 0 an 255. 
Please clarify.

xobs, 2018-03-02:
`::analogRead()` returns a value between 0 and 1023 inclusive.  The OS-level implementation is at https://github.com/chibitronics/ltc-os/blob/master/orchard/arduino-io.c#L412, and you can see `::analogReadResolution()` can be used to increase the resolution to up to 12 bits (0..4095).

`::analogWrite()` also mimics Arduino's implementation, and you're right the range is different.  The `::analogWrite()` function expects a value between (0..255).

samelhusseini, 2018-03-02:
Makes perfect sense. I think were we went wrong was in assuming the underlying read and write had the same ranges. 
Now that we know the ranges of the underlying APIs. What should we expose to users. 
We've already agreed that the high level 'read level' and 'set level' APIs should have a range of 0-100 as per Natalie's request, but what should the ranges of the lower level APIs be.
I see two options here, we stay consistent with the OS and give the user the same granularity, (0..1023) for read (0.255) for write. or we try and make the two APIs consistent, mapping levels to one of the two ranges.
Thoughts? @xobs @nataliefreed @pelikhan

nataliefreed, 2018-03-02:
My two cents:
read level and set level 0-100 for beginner-level simplicity.
analogRead and analogWrite 0-1023 and 0-255 to match Arduino conventions, since those functions match Arduino syntax, and also to have access to higher resolution reads/writes.
Checking in with @jieq as well.

pelikhan, 2018-03-02:
We should align with MakeCode which is 0..1023 both on read/write.

samelhusseini, 2018-03-02:
@nataliefreed are you also okay if I change the top level read block to say "read level" instead of "read analog", in order to make it consistent with "set level" (for analog).

PR #192 (2018-03-30): Fix analog read range assumption of (0.1023). 
Fixes #191 

change "read analog" block to "read level"
read level (analog) block now returns values between 0 and 100.
commit e3a8254e (2018-03-30): Fix analog read range assumption of (0.1023).  (#192)

* Fix analog read range assumption of (0.1023). Fixes #191

* Make the fix in pins.cpp instead of shims

* Fix sensing namespace name
closed 2018-03-30 by samelhusseini