microsoft/pxt-chibitronics #210: There is no closing button when opening examples under "Tutorials", "Guides" and "Advanced"

Helen09, 2018-05-17:
**Repro steps:**
• Open https://makecode.mindstorms.com/beta
• Navigate to "Tutorials", "Examples", "Love to Code", "Guides", "Advanced" and open examples under them

**Expected result:**
Every opened example should have a closing button on the opened window
[image]

**Actual result:**
 There is no closing button when opening examples under "Tutorials", "Guides" and "Advanced"
[image]

samelhusseini, 2018-05-18:
It's there, but it's white. We need to think through a fix for this.

Helen09, 2018-06-13:
Closed as fixed

closed 2018-06-13 by Helen09