microsoft/pxt-chibitronics #213: The color on simulator is different with the color choice on content pane

Helen09, 2018-05-18:
**Repro Steps:**
• Open https://makecode.mindstorms.com/beta
• Navigate to "Advanced" and click "Simple Color" to open this example
• After loading this project, click every different color and observe the simulator
**Expected result:**
The color on simulator should be same as the color choice on content pane 
**Actual result:**
The color on simulator is different with the color choice on content pane 
[image]

samelhusseini, 2018-05-18:
Black is off not the color black.

samelhusseini, 2018-05-18:
Maybe we should use OFF in the text.

Helen09, 2018-05-30:
When selecting white, it shows as below
[image]

When selecting pink, it shows as below
[image]