microsoft/pxt-chibitronics #234: Sensing blocks and animations block don't work together

samelhusseini, 2018-06-22:
It appears using the sensing blocks to trigger animations on a different pin crashes the running code. 
Here's an example: 

```
sensing.onPinEvent(DigitalPin.D0, PinEvent.Pressed, function () {
    rgb.setColor(rgb.wheel(Math.randomRange(0, 255)))
    for (let i = 0; i < 500; i++) {
        lights.effect(Effect.Twinkle, AnalogPin.A2)
    }
})
```

I suspect it's an issue with the thread that the sensing block starts, maybe that's interfering somehow. 
@xobs any insight here. 

This issue was reported by @nataliefreed and Ryan Jenkins.

-------------------

and here's an example where the sensing block works: 
```
sensing.onPinEvent(DigitalPin.D0, PinEvent.Pressed, function () {
    rgb.setColor(rgb.wheel(Math.randomRange(0, 255)))
    lights.set(Math.randomRange(1, 6), Math.randomRange(0, 2))
})
```

xobs, 2018-09-06:
The board enters an error condition, as evidenced by the blinking red LED.  Investigating now...

It's actually crashed in the timer handler:

```
(gdb) bt
#0  0x00001cf4 in errorCondition () at esplanade_app.c:46
#1  0x00001736 in HardFault_Handler_C (context=<optimized out>, is_irq=<optimized out>) at orchard-vectors.c:40
#2  <signal handler called>
#3  0x00002400 in chVTDoSetI (vtp=<optimized out>, delay=<optimized out>, vtfunc=<optimized out>,
    par=<optimized out>) at ../os/rt/src/chvt.c:167
#4  0x0000251c in chSchGoSleepTimeoutS (newstate=newstate@entry=8 '\b', time=<optimized out>)
    at ../os/rt/src/chschd.c:338
#5  0x000026ec in chThdSleepS (time=<optimized out>) at ../os/rt/include/chthreads.h:268
#6  chThdSleep (time=<optimized out>) at ../os/rt/src/chthreads.c:314
#7  0x00005324 in delay (msecs=<optimized out>) at arduino-time.c:17
#8  0x00005d16 in loops::pause (ms=<optimized out>) at src\core/basic.cpp:52
#9  0x000067aa in ?? ()
(gdb)
```

xobs, 2018-09-06:
Fixed in ca0f5742bd7d80c777191da4bfa6ca42d083c1b7 by increasing the stack size.

jieq, 2018-09-10:
When I try using the sensing pin and twinkle effect (one of the most
popular combos in workshops), still seems to crash causing program light to
blink red.

Thanks for looking into this!

xobs, 2018-09-11:
This fix is sitting in the `fix-twinkle-effect` branch and hasn't been merged yet, so it's not live.  You'd need to check out and build that branch yourself.

xobs, 2018-09-14:
It should be in beta now.  @jieq can you try http://makecode.chibitronics.com/beta ?

jieq, 2018-09-14:
Huzzah!  Works great now :)  Thanks!

--Jie